sparkfun/OpenLog_Artemis_GNSS_Logger #2: Max sample recording rate?

seanmcleod, 2020-07-15:
What is the maximum sample recording rate for logging raw UBX messages to the SD card for later processing with RTKLib etc.?

Can it achieve 20-25Hz when connected to the ZED-F9P?

Regards

adamgarbo, 2020-07-15:
Hi @seanmcleod,

Short answer is, it depends. 

If you look here: https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/c22cb286ec099bf7a5f8bb897687ceb3cba943ef/Firmware/OpenLog_Artemis_GNSS_Logging/Sensors.ino#L254

You'll see that above a sampling rate of 5 Hz, all constellations except GPS are turned off in order to maintain communications over I2C. Below 5 Hz, you can sample all constellations. 

To log RAWX data from all constellations at 20 Hz, you'd likely need do so directly over Serial.

Cheers,
Adam

seanmcleod, 2020-07-15:
Is there a formula for the size of RAWX data for GPS constellation only to compare against the maximum I2C bandwidth?

In other words if we were happy with GPS constellation only can we then get 20Hz?

Or would we still exceed the I2C bandwidth and we could for example only get 12Hz?

adamgarbo, 2020-07-15:
To my knowledge, u-blox doesn't make this information readily available. I've tried looking for it in the past in order to calculate data budgets, but have never been able to find any definitive numbers. While it's ultimately dependent on the number of satellites you're tracking, there must be an upper limit to the amount of data per band that can be received.

I don't believe GPS @ 20 Hz should be an issue for the GNSS logger. If you'd like GPS + GLONASS above 5 Hz, I'd try Serial or you can [modify the firmware](https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/c22cb286ec099bf7a5f8bb897687ceb3cba943ef/Firmware/OpenLog_Artemis_GNSS_Logging/Sensors.ino#L257-L260) to override the 5 Hz and test your luck.

Up here in Canada, I usually rely on GPS + GLONASS, since those are the two supported constellations for [NRCan's PPP tool](https://www.nrcan.gc.ca/maps-tools-publications/tools/geodetic-reference-systems-tools/tools-applications/10925#ppp), though I'm usually sampling below 5 Hz.

Cheers,
Adam

seanmcleod, 2020-07-15:
Ah, I see the following code just a bit above.

https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/c22cb286ec099bf7a5f8bb897687ceb3cba943ef/Firmware/OpenLog_Artemis_GNSS_Logging/Sensors.ino#L202-L206

Do these limits take into account the I2C bandwidth, or do they just refer to the Zed-9P's limits in terms of what it can output?

For example I see it mentions 8Hz for all constellations, but the code @adamgarbo mentioned limits it to 5Hz, is it just being more conservative or is that due to the I2C bandwidth limits?

adamgarbo, 2020-07-15:
@seanmcleod,

While the limits are specific to the ZED-F9P receiver (see below), I believe @PaulZC implemented the 5 Hz limit based on the results of his own I2C tests with the receiver. I'm sure he'll chime in here, eventually! 

[image]

seanmcleod, 2020-07-15:
Thanks @adamgarbo for the data. Looks like a bit of trial and error required once I get the units.

We're planning on using it in an aircraft to measure landing distances etc. hence the reason for ideally 20-25Hz recording rate for the RAW data.

PaulZC, 2020-07-15:
Hi @seanmcleod ,

Like @adamgarbo says, there is indeed a bit of trial and error involved.

The ZED is capable of delivering RAW navigation solutions at up to 25Hz depending on how many constellations are being tracked. But if you're using RTK, you're limited to 20Hz / 8Hz. The 5Hz comment is from a previous version of the code and is out of date - I need to remove that.

However, there is also the I2C bandwidth to throw into the mix. Depending on how many SVs are being tracked, a RAWX packet can be approx. ~2KB in length. You're going to want to log the SFRBX messages too which can add another 54 bytes per SV. There's no way to tell the ZED that we promise to only use 400kHz I<sup>2</sup>C, so _I think_ it plays safe and drops the navigation rate once the selected messages and number of constellations threaten to overload the I2C interface at 100kHz.

Like Adam says, if you really want to push it to the max, you'll need to log serial data instead.

I'm going to close this as this isn't really an issue with the code. If you want to discuss further, please just drop me a line. You'll find my email on my profile.

Very best wishes,

Paul

closed 2020-07-15 by PaulZC

PaulZC, 2020-07-15:
Hi @seanmcleod ,
I've just tried pushing the ZED-F9P to 25Hz while logging RAWX and SFRBX messages - and it does work.
Caveats:
- This is on GPS only.
- I've also disabled the USB and UART ports to help lighten the ZED's load. (You can do that through the menus.)
```
2020/07/15 22:03:29.99 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.03 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.04 UBX Class:0x2 ID:0x13 RXM-SFRBX
2020/07/15 22:03:30.05 UBX Class:0x2 ID:0x13 RXM-SFRBX
2020/07/15 22:03:30.05 UBX Class:0x2 ID:0x13 RXM-SFRBX
2020/07/15 22:03:30.05 UBX Class:0x2 ID:0x13 RXM-SFRBX
2020/07/15 22:03:30.05 UBX Class:0x2 ID:0x13 RXM-SFRBX
2020/07/15 22:03:30.06 UBX Class:0x2 ID:0x13 RXM-SFRBX
2020/07/15 22:03:30.06 UBX Class:0x2 ID:0x13 RXM-SFRBX
2020/07/15 22:03:30.06 UBX Class:0x2 ID:0x13 RXM-SFRBX
2020/07/15 22:03:30.06 UBX Class:0x2 ID:0x13 RXM-SFRBX
2020/07/15 22:03:30.07 UBX Class:0x2 ID:0x13 RXM-SFRBX
2020/07/15 22:03:30.09 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.12 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.16 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.19 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.23 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.26 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.32 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.37 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.40 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.44 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.47 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.51 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.56 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.59 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.63 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.66 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.72 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.75 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.79 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.87 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.89 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.94 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:30.97 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:31.01 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:31.04 UBX Class:0x2 ID:0x15 RXM-RAWX
2020/07/15 22:03:31.07 UBX Class:0x2 ID:0x15 RXM-RAWX
```
All the best,
Paul

seanmcleod, 2020-07-15:
@PaulZC excellent, thanks for testing!

One last/another question, will the following, taken from your Adafruit logger, also work currently with the Artemis logger?

> You can connect a push switch between GND and the SparkFun INT pin. Pushing it will generate a TIM_TM2 message which will get logged with the RAWX data. RTKLIB can be used to export these events.

PaulZC, 2020-07-15:
Hi @seanmcleod ,
Short answer - yes.  Hook your switch or 3.3V logic signal to the INT pin on the ZED breakout, enable the TIM-TM2 messages through the menus and you're good to go.
BTW: here is [more information about the 5Hz limit](https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/master/Firmware/OpenLog_Artemis_GNSS_Logging/settings.h#L38-L43). At some point I will need to set ```minMeasIntervalRAWXAll``` by module type.
All the best,
Paul