sparkfun/OpenLog_Artemis_GNSS_Logger #8: Raw data logging comparison: GNSS Logger vs. OLA

adamgarbo, 2020-08-23:
Hi there,

I'm extremely interested in the use of the OLA as a high-precision GNSS data logger. The OLA and the GPS-RTK2 ZED-F9P boards make an amazing combination. @PaulZC's GNSS Logger code has also really made logging data from the ZED-F9P a breeze.

### Background
I've been curious about how the GNSS Logger firmware, which records raw UBX data over I2C, would compare to the default OLA firmware and logging UBX data over serial. To this end, I've been conducting a series of tests comparing the two firmware. I thought I'd share my results and thoughts below to help provide information to other potential users and to identify ways to improve the firmware used to perform this type of data logging.

### Methods: Testing setup
* Both systems used identical components, as shown in Table 1. 
* The main difference between the two systems was that the OLA was connected directly via serial using the 0.1" Tx/Rx/3V/GND header pins, while the GNSS Logger was connected via I2C (Qwiic connector cable).
* The test was conducted outdoors over 24 hours.

**Table 1.** Components used for the comparison test.
Components: |
-- |
SparkFun OpenLog Artemis (shiny new red version) 
SparkFun GPS-RTK2 Board ZED-F9P 
u-blox ANN-MB antenna 

**Table 2.** Firmware and hardware used with the OLAs, and rates and protocols used to record data.
  | Firmware | Hardware | Protocol | Frequency
-- | -- | -- | -- | -- 
**OLA** | v15 | v10 | Serial | 1 Hz
**GNSS Logger** | v12 | v10 | I2C | 1 Hz

**Table 3.** Enabled constellations and recorded UBX messages.
Constellations | UBX Messages
-- | --
GPS | RAWX
GLONASS | SFRBX

### Methods: Data processing
* Data was downloaded and processed using [RTKLIB demo5 b33c](http://rtkexplorer.com/downloads/rtklib-code/).
* Hourly log files were combined and processed to create RINEX files for each recording duration (i.e. 1 hour, 2 hours, 3 hours, etc.)
* RINEX files were submitted to [NRCan's Precise Point Positioning (PPP) tool](https://webapp.geod.nrcan.gc.ca/geod/tools-outils/ppp.php?locale=en) 
* GPS & GLONASS	NRCan Rapid results are displayed in Tables 4-5 and Figures 1-2.

### Results
* The OLA collected data for the entire 24 hours duration.
* The GNSS Logger randomly stopped logging data after 5.5 hours.
  * Note: This bug has previously been observed with the v4 hardware revision of the OLA.
* It's important to note that the accuracy results are based on NRCan "Rapid" ephemerides.
  * Once the "Final" ephemerides are available (in 13-15 days) the accuracy would likely increase.

**Table 4.** GNSS Logger/OLA RTKLIB processed observations, navigation messages and errors.
Duration (h) | Observations | NAV Messages | Errors | Latitude (m) | Longitude (m) | Height (m)
-- | -- | -- | -- | -- | -- | --
1 | 3111 | 31 | 2 | 0.073 | 0.156 | 0.178
2 | 6225 | 64 | 3 | 0.049 | 0.140 | 0.089
3 | 9352 | 89 | 4 | 0.026 | 0.060 | 0.078
4 | 12564 | 117 | 4 | 0.024 | 0.054 | 0.060
5 | 15710 | 140 | 4 | 0.022 | 0.051 | 0.059

**Table 5.** GNSS Logger/OLA NRCan PPP accuracy results (± m).
Duration (h) | Observations | NAV Messages | Errors | Latitude (m) | Longitude (m) | Height (m)
-- | -- | -- | -- | -- | -- | --
1 | 3538 | 30 | 48 | 0.065 | 0.140 | 0.195
2 | 7079 | 55 | 99 | 0.045 | 0.106 | 0.087
3 | 10624 | 88 | 150 | 0.022 | 0.046 | 0.073
4 | 14159 | 108 | 213 | 0.019 | 0.040 | 0.058
5 | 17691 | 139 | 263 | 0.017 | 0.038 | 0.054

[image]
**Figure 1.** GNSS Logger/OLA RTKLIB processed observations, navigation messages and errors.

[image]
**Figure 2.** GNSS Logger/OLA NRCan PPP accuracy results (± m).

### Conclusions
* While it was a short duration test, both the GNSS Logger and OLA produced impressive results.

#### OLA Pros:
* The OLA records more observations over serial than the GNSS Logger over I2C.
* Initial PPP processing results indicate greater accuracy can be achieved by logging over serial. 
* The OLA concurrently logs battery voltage, IMU and other sensor data to a separate TXT file over I2C.
* Can log GPS/GLONASS data in theory at >20 Hz.
#### OLA Cons:
* No RTC synchronization.
* Serial logging provides the user with very little control over the GNSS receiver.
  * This necessitates the use of u-blox u-center software to configure the GNSS receiver.

#### GNSS Logger Pros:
* The GNSS Logger does a good job of rejecting errors.
* Provides greater control over the configuration of the GNSS receiver through menus over the Serial Monitor.
* Allows for continuous RTC date/time synchronization.
#### GNSS Logger Cons:
* I2C logging appears to be less accurate than serial.
* Does not record other measurements to a separate log file (e.g. battery voltage, IMU data, etc.)
* Limited to recording GPS/GLONASS at 4 Hz.
* A bug is causing the GNSS Logger's logging to randomly fail and not recover, which needs to be investigated.

### Next Steps
* Identify what is causing the GNSS Logger to randomly stop logging.
* Attempt a second 24-hour test.
* Look into adding functionality to allow the GNSS Logger to record battery voltage and other sensor measurements.

Happy to receive any feedback and/or suggestions!

Cheers,
Adam

nseidle, 2020-08-25:
Hi Adam - This is incredible and much appreciated. Thank you very much for putting the OLA firmwares through the paces. We'll investigate the the GNSS Logger stopping after a few hours.

PaulZC, 2020-08-25:
Hi Adam.
This is great stuff - thank you!
I'll start investigating why the GNSS logger stops logging today.
I'd also like to dig into the number of errors seen in the OLA data. Can you let me know:
- What baud rate were you using for the serial logging?
- Was the OLA logging IMU data at the same time (or did you have that disabled)?
I wonder if you can also do a little detective work for me? Can you please play back some of your logged data through u-center (see below). If you open Messages View, do you see any UNKNOWN messages from either system (OLA vs GNSS)? I'm trying to figure out what those errors actually are: dropped bytes, checksum failures, etc..
Thanks again!
All the best,
Paul
[image]

adamgarbo, 2020-08-25:
Hi Paul,

Both the OLA and the ZED-F9P were configured to communicate at a baud rate of 230400. During the test, the OLA was also logging all default variables (date, time, IMU, voltage, etc.)

Moving on to the detective work, I opened a .ubx file that RTKLIB identified as having 48 errors in u-center. I played back the data (1 hour) but did not observe any UKNOWN messages. The only messages that were observed were the RXM-UBX and RXM-SFRBX. 

I believe what we may be observing is simply incomplete UBX frames. I recall speaking with Tim Everett about this and he advised that RTKLIB should be tolerant incomplete data. He suggested a good way to verify this was to edit a .ubx file with a binary editor to generate a few incomplete messages. I tested this by making four random changes to synchronization characters (0xB5 0x62) and the RTKLIB error count increased from 48 to 52.

I am currently testing both the v04 and v10 hardware revisions with the GNSS Logger firmware. The test has been running over the past 16 hours and both are still confirmed to be logging. Will keep you posted!

Cheers,
Adam

adamgarbo, 2020-08-26:
I've just completed another 24-hour test. This one compared the v04 and v10 hardware revisions against one another. Both OLAs were running the v12 GNSS Logger firmware. 

I've just begun processing the results and will update more later, but it was interesting to see how the file sizes differed. We'll see if it was indicative of anything once the data gets processed by RTKLIB. It is possible it was environmental, though the testing setup was identical. 

It was also good to see that both boards completed the test, but still puzzling why there were failures in the past.

[image]

PaulZC, 2020-08-26:
Many thanks Adam,

I'm still digging into this...

I'm confused by the difference in the number of observations. I shall keep scratching my head about that one.

When you see the GNSS Logger stop logging after ~5.5 hours, are you sure that the logging actually stops?
Does the STATUS LED stop blinking?
Can you still open the menu when it is in this state?
I'm trying to figure out if: the code is crashing; or if the SD-writing is stalling; or if data is still being logged but the file ends up containing an error that RTKLIB/NRCan cannot deal with?

If you have time, can you try running some of your bad files through this tool:
https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/tree/master/Utils
It will hopefully let us identify if bytes are being dropped, corrupted, or if I2C errors are introducing extra bytes. Please post some results and we'll look for a smoking gun! If you really want to get stuck in, you could look at the raw data in the files with a hex editor and have a look at the problematic locations identified by the tool.

I ran an overnight test last night, logging RAWX and SFRBX at 1Hz from all constellations. The GNSS Logger code includes a 24K RingBuffer which is used to store the data between reading it from I2C and writing it to SD card. The SD card does occasionally throw a wobbler and writing stalls for the best part of a second. I'm convinced this is integral to the SD card. I suspect it is shuffling data around to avoid pounding one area too hard, or updating the file allocation table takes longer than usual. Anyway, the buffer needs to be big enough to store all of the I2C data while the SD card sorts itself out. During last night's test I did see bufAvail hit its limit: ```03:06:52.980 -> storeData: Max bufAvail: 24575```. So, unsurprisingly, the logged data does lose sync. But it did recover and keep logging afterwards...

Thanks again!

All the best,

Paul

adamgarbo, 2020-08-26:
Hi Paul,

**GNSS Logger (I2C)**

When I halted the OLA vs GNSS Logger test, I confirmed that there were no STAT or TX lights flashing on the GNSS Logger. However, its ZED-F9P was still happily blinking away. Also, the log files simply stopped being created after a certain time. As it wasn't connected via serial, I'm afraid I wasn't able to see if there were any error messages. 

I did process the GNSS Logger's final .ubx file with the handy Python tool you wrote:

```
UBX Integrity Checker

Enter the UBX filename (default: ./20200822_153857.ubx): 20200822_210002.ubx
Could this file contain any NMEA messages? (Y/n): n

Processing 20200822_210002.ubx

UBX Sync Char 1 (0xB5) found at byte 0. Checking for Sync Char 2
Panic!! UBX checksum error!
Sync lost at byte 1758039. Attemting to re-sync.
Sync has been lost. Currently processing byte 1758039. Rewinding to byte 1756293
Warning: * found at byte 1758321! Are you sure this file does not contain NMEA messages?
Warning: * found at byte 1758378! Are you sure this file does not contain NMEA messages?
UBX Sync Char 1 (0xB5) found at byte 1758389. Checking for Sync Char 2
Panic!! Was expecting Sync Char 2 (0x62) but did not receive one!
Sync lost at byte 1758390. Attemting to re-sync
UBX Sync Char 1 (0xB5) found at byte 1758974. Checking for Sync Char 2
Sync successfully re-established at byte 1760725. The UBX message started at byte 1758974

Processed 5187271 bytes
File size was 5187271
Longest UBX message was 1936 data bytes
Message types and totals were:
Message type: 0x02 0x13   Total: 38612
Message type: 0x01 0x21   Total: 2248
Message type: 0x02 0x15   Total: 1925
Number of successful resyncs: 1
```

**OpenLog Artemis (Serial)**

Please see below for the output of the UBX Integrity Checker for the .ubx serial log file that RTKLIB identified as having 48 errors:

```
Enter the UBX filename (default: ./20200823_150000.ubx): 20200822_160000.ubx
Could this file contain any NMEA messages? (Y/n): n

Processing 20200822_160000.ubx

UBX Sync Char 1 (0xB5) found at byte 0. Checking for Sync Char 2
Panic!! UBX checksum error!
Sync lost at byte 60775. Attemting to re-sync.
Sync has been lost. Currently processing byte 60775. Rewinding to byte 59701
UBX Sync Char 1 (0xB5) found at byte 60868. Checking for Sync Char 2
Panic!! Was expecting Sync Char 2 (0x62) but did not receive one!
Sync lost at byte 60869. Attemting to re-sync
Warning: * found at byte 61425! Are you sure this file does not contain NMEA messages?
Warning: * found at byte 61761! Are you sure this file does not contain NMEA messages?
UBX Sync Char 1 (0xB5) found at byte 61849. Checking for Sync Char 2
Sync successfully re-established at byte 61880. The UBX message started at byte 61849

Panic!! UBX checksum error!
Sync lost at byte 66680. Attemting to re-sync.
Sync has been lost. Currently processing byte 66680. Rewinding to byte 65606
UBX Sync Char 1 (0xB5) found at byte 66778. Checking for Sync Char 2
Panic!! Was expecting Sync Char 2 (0x62) but did not receive one!
Sync lost at byte 66779. Attemting to re-sync
Warning: * found at byte 66919! Are you sure this file does not contain NMEA messages?
UBX Sync Char 1 (0xB5) found at byte 67160. Checking for Sync Char 2
Panic!! Was expecting Sync Char 2 (0x62) but did not receive one!
Sync lost at byte 67161. Attemting to re-sync
[... 81 lines cut ...]

Panic!! UBX checksum error!
Sync lost at byte 730894. Attemting to re-sync.
Sync has been lost. Currently processing byte 730894. Rewinding to byte 730868
UBX Sync Char 1 (0xB5) found at byte 730901. Checking for Sync Char 2
Sync successfully re-established at byte 730932. The UBX message started at byte 730901

Panic!! UBX checksum error!
Sync lost at byte 733924. Attemting to re-sync.
Panic! Maximum rewind attempts reached! Aborting...

Processed 733925 bytes
File size was 5122737
FILE SIZE MISMATCH!!
Longest UBX message was 1136 data bytes
Message types and totals were:
Message type: 0x02 0x15   Total: 486
Message type: 0x02 0x13   Total: 5135
Number of successful resyncs: 10

Bye!
```

Cheers,
Adam

PaulZC, 2020-08-26:
Nice one Adam - thank you,

If we were dropping bytes, due to serial receive buffer overflows, then I think we might expect to see sync being re-established earlier then when sync was lost. The ```message started at byte``` values would, I think, be smaller than the ```Sync lost at byte``` values. But we don't. ```message started at``` is always larger than ```sync lost at```.

If it was data corruption, then we'd expect to see sync re-established immediately after sync was lost due to a checksum failure. But we don't see that either.

What we do see is the ```Sync lost at byte``` values being consistently higher than the ```message started at byte``` values by quite a bit. Sometimes a KB or more. Which is making me think that we are consistently losing the two start of frame chars in each of these events...

Maybe a buffer overflow event does explain this, but it is an overflow that causes the start of the following packet to be lost...

I need to think more about this.

I will try adding an extra serial RX RingBuffer to the OLA code, to hold a glut of serial data, to see if the majority of those errors go away.

More later!

adamgarbo, 2020-08-26:
Thanks, Pau!

Did you also have any thoughts as to why the GNSS Logger is recording significantly fewer observations over I2C than the OLA is over serial at the same rate (1 Hz)?

I'm happy to conduct any testing configuration that you suggest!

Cheers,
Adam

PaulZC, 2020-08-26:
Hi Adam,

OK. I think I understand why we see occasional errors in the GNSS Logger data. I think it is just the 24K buffer overflowing when: there's lots of RAWX data coming in; and the SD card has one of its occasional hissy fits (when it takes a second or more to finish writing). You could check if making the buffer larger makes the errors go away:
https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/master/Firmware/OpenLog_Artemis_GNSS_Logging/storeData.ino#L28
If you have time, please try increasing it to 32K or 48K and see if the errors go away.

The u-blox chips are clever beasties. I'm beginning to suspect that they throttle back on the SFRBX messages when the I2C bus is starting to look congested. One way to check would be to look at the number of SFRBX (0x02 0x13) messages vs the number of RAWX (0x02 0x15) on serial vs. I2C. The Python tool will give you a count of each type.

I think I also understand the errors in the OLA data too. Let's continue that discussion over here:
https://github.com/sparkfun/OpenLog_Artemis/issues/30

Thanks!
Paul

adamgarbo, 2020-08-28:
Hi Paul,

I'll definitely try increasing the RingBuffer to see if it eliminates the random errors.

The more pressing issue I'm encountering is with the GNSS Logger logging failing. It occurred again this evening after only an hour of logging. The PWR LED and the GPS-RTK2 board LEDS were on, but again no STAT or TX LED activity. 

I'm running a test overnight with debug messages enabled and the serial output logged via CoolTerm to see if we can figure out what's actually happening when it stops logging. So far, the failures have occurred after 41 hours, 5 hours and 1 hour. So random!

Cheers,
Adam

adamgarbo, 2020-08-28:
Ah ha!

```
storeData: UBXbuffer overflow! You need to increase the size of UBXbufferSize in storeData. Freezing...
```

PaulZC, 2020-08-28:
Hi Adam,

Fantastic. Thanks for finding this! I _think_ this means we have received a legitimate UBX frame with a length greater than 4KB... That _should_ be impossible. [Scratches head.]

It would be great to trap this, just to see if we are receiving some kind of 'jumbo frame'. But that would mean adding extra debug prints to trap the ```ubx_length```, ```ubx_class``` and ```ubx_ID```.

Going forward, we can either increase ```UBXbufferSize``` to maybe 8KB and see if the problem goes away.
Or, as life is short, we can simply discard the data.

To discard the data, we'll need to change lines 223 to 225: from:

```
Serial.print(F("storeData: UBXbuffer overflow! You need to increase the size of UBXbufferSize in storeData. Freezing..."));
while(1)
;
```

to:

```
Serial.print(F("storeData: UBXbuffer overflow! Re-syncing..."));
ubx_state = sync_lost; // Discard the data and force a re-sync
```

At least this will let the code keep logging if it does gets a 'jumbo packet'.

Half of me would really like to dig into this properly and fully understand what is happening. But the other half recognizes that we are heading down the rabbit hole and is telling me to just discard the data...

I shall sit on the fence over the weekend...

Thanks again for finding this!

Paul

adamgarbo, 2020-08-29:
Hi Paul,

To test the UBXbuffer overflow issue, I conducted a 12-hour test with two GNSS Loggers configured with `UBXbufferSize = 8192`. I also made use of the PWR LED to provide a visual indication of whether an overflow occurred.

```
if (UBXpointer == UBXbufferSize) //This should never happen!
{
  Serial.print(F("storeData: UBXbuffer overflow! Re-syncing..."));
  ubx_state = sync_lost; // Discard the data and force a re-sync
  powerLEDOn(); //Visual indication of UBXbuffer overflow
}
```

During recovery of the instruments, I observed that one GNSS Logger had the PWR LED turned on, while the other did not. Processing the .ubx files with UBX_Integrirty_Checker.py did not reveal any sync losses. Processing the .ubx files with RTKLIB indicated that the logger with the illuminated PWR LED had a single error, while the other had none.

So far, it appears that increasing the size of the UBXbuffer and discarding any UBX frames that are larger than 8 KB is an adequate fix for this issue. I agree that we may be going down a rabbit hole.

Cheers,
Adam

[image]

[image]

adamgarbo, 2020-08-29:
Hi Paul,

In the most recent test I performed to investigate the UBXbuffer overflow issue, I was actually concurrently testing three OLA and ZED-F9P receivers. 

To follow-up on a separate issue identified earlier, the OLA recording UBX data over serial is consistently recording a significantly greater number of observations than the GNSS Loggers over I2C. Have you given any thought as to why the GNSS Loggers are missing observations? The testing configurations and results are shown below.

Cheers,
Adam

**Table 1.** Testing configurations.
Version | Hardware | Firmware | Receiver | Antenna | Comments
-- | -- | -- | -- | -- | --
gnss_1 | OLA v10 | GNSS_Logger v12 | GPS-RTK2 | u-blox ANN-MB-00
gnss_2 | OLA v10 | GNSS_Logger v12 | GPS-RTK-SMA | Tallysman HC872
ola | OLA v10 | OLA v15 | GPS-RTK2 | u-blox ANN-MB-00 | IMU disabled. Date, time and voltage logged at 1 Hz.

**Table 2.** RTKLIB processing results.  
Version | Observations | NAV Messages | Errors
-- | -- | -- | --
gnss_1 | 46926 | 410 | 0
gnss_2 | 47517 | 399 | 1
ola | 53820 | 1042 | 8

[image]
**Figure 1.** File sizes by hour.

PaulZC, 2020-08-29:
Hi @adamgarbo ,

Many thanks for this.

I don't think the GNSS Logger is 'missing' observations. Instead, I _suspect_ the module is throttling the data flow through I2C so that it doesn't overload the interface. It would be really interesting to repeat the comparison but change the serial baud rate to 115200 - which would be a closer match to I2C running at 100kHz. I wonder if there would then be a closer match in the number of observations?

Thanks again,

Paul

adamgarbo, 2020-08-31:
Hi Paul,

Instead of decreasing the serial baud rate, I decided to increase the I2C frequency up to 400 kHz and see what would happen. Turns out that the files become much more similar in size. In fact, the GNSS Logger looks to be collecting more data at some points.

I haven't examined whether there are any errors in the data collected by the GNSS Logger and will update this post later once I've processed them with RTKLIB.

[image]
**Figure 1.** OLA (Serial @ 230400) and GNSS Logger (I2C @ 400 kHz) collected data log file sizes.

Cheers,
Adam

PaulZC, 2020-08-31:
Nice one Adam - thank you!

Glad my hunch about the data-throttling seems to be correct!

All the best,

Paul

PaulZC, 2020-12-07:
Hi Adam (@adamgarbo ),
I've kind of lost track as to where we ended up with this issue...
Is there still something that needs investigating?
Thanks!
Paul

adamgarbo, 2020-12-08:
Hi Paul,

This was never meant to be an issue, though we were able to identify and resolve a number of them as a result. It was more of an exploration of the potential of the OLA as a GNSS logger.

The main takeaways have been that in order to log I2C data at an equivalent rate to serial (e.g. 230400 bps), a bus speed of 400 kHz is required. To date, all of the tests I've performed at 400 kHz have worked well when only using the Artemis' pull-ups. At 100 kHz, as we identified, the ZED-F9P throttles data to the I2C bus and fewer observations are collected. I'm uncertain regarding the long-term stability of a 400 kHz bus speed with no pull-ups. Maybe you can comment?

I'm happy for the issue to either be left open or closed. Either works for me! I do tend to reference this issue quite frequently, so it may be worthwhile for me to write up some documentation, pulling from the experiences outlined above.

Cheers,
Adam

PaulZC, 2020-12-08:
OK - thanks Adam,
I'll close this for now. We can always re-open if required.
All the best,
Paul

closed 2020-12-08 by PaulZC