sparkfun/OpenLog_Artemis_GNSS_Logger #14: Feature request: File access and creation timestamps

adamgarbo, 2020-09-07:
Hi @PaulZC,

Just a reminder to bring over the file access and creation timestamp changes from the OpenLog Artemis to the GNSS Logger repository.

In case you're interested, in my fork of the GNSS Logger I've added functionality to synchronize the RTC with the GNSS before creating the first log file.

https://github.com/adamgarbo/OpenLog_Artemis_GNSS_Logger/blob/544f335361b8d0989313de86934e69299137a0cc/Firmware/OpenLog_Artemis_GNSS_Logging/OpenLog_Artemis_GNSS_Logging.ino#L535-L588

Granted, this is more specific to my needs as I've adopted a log file naming convention based on the date and time. It might be worth giving some thought to providing this as an option for the OLA/GNSS Logger. I could see this naming convention being a desirable alternative for many users to the `dataLogXXXX.ubx`, `serialLogXXXX.txt`.

https://github.com/adamgarbo/OpenLog_Artemis_GNSS_Logger/blob/544f335361b8d0989313de86934e69299137a0cc/Firmware/OpenLog_Artemis_GNSS_Logging/OpenLog_Artemis_GNSS_Logging.ino#L418-L424

Cheers,
Adam

PaulZC, 2020-12-07:
Added in [this commit](https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/commit/be1f817c9daf4193554f8a947587673db41e3077).
I'll leave this open until v1.3 is released.

PaulZC, 2020-12-08:
v1.3 has just gone live. Please kick the tyres when time permits!
All the best,
Paul

closed 2020-12-08 by PaulZC