sparkfun/OpenLog_Artemis_GNSS_Logger #15: Feature request: Diagnostics log file

adamgarbo, 2020-10-18:
Having recently picked back up my GNSS Loggers, I've found myself wishing I had a little more information about how the OpenLog Artemis is performing during testing.

While the main goal of the GNSS Logger has always been to record UBX data, I believe that it could be a worthwhile addition to enable the periodic recording of specific diagnostic information to a separate log file. 

For example, users (myself included) would likely be interested in being able to track their battery voltage in order to assess the performance of their equipment over extended periods of time. While we've identified that simultaneous logging of serial data from a GNSS receiver and I2C data from the OLA's onboard sensors [does not work as well as hoped](https://github.com/sparkfun/OpenLog_Artemis/issues/30), my thought was the diagnostic information could be measured very intermittently (i.e. hourly intervals).

Cheers,
Adam

PaulZC, 2020-12-07:
Hi Adam (@adamgarbo ),
I'm struggling to think of a 'clean' way of doing this - which doesn't need additional 'timers' and which doesn't run the risk of stalling fast RAWX logging...
The best I've come up with is to add any housekeeping data to a dedicated file when the Logger goes to sleep. I.e.:
- sync GNSS data file
- close GNSS data file
- open housekeeping data file, update it, close it
- go to sleep

Does that sound adequate?
It doesn't cover the "continuous logging (until the battery goes flat)" case of course...
Thanks,
Paul

PaulZC, 2020-12-08:
Of course, you could always use a _second_ OLA to log the battery voltage of the GNSS Logger?!

adamgarbo, 2020-12-08:
Hi @PaulZC,
> Of course, you could always use a _second_ OLA to log the battery voltage of the GNSS Logger?!

Hah! Can you imagine! 

I've actually been giving this some thought as well. At slower logging rates (e.g. 1 Hz), I expect that there would be enough processing time available to periodically write to a separate log file, but at higher rates, I expect it would become more difficult to interject this kind of operation.

In past projects, I've used Rob Tillart's Statistic library (https://github.com/RobTillaart/Statistic) to keep a running calculation of battery voltage min/max/mean values that are logged and transmitted over Iridium. Minimum battery voltage is of course the most telling about the health of a system. It could be that a similar approach could work well here.

I'll give this some more thought!

Cheers,
Adam

PaulZC, 2020-12-08:
Thanks Adam,
I'll leave this open for now.
All the best,
Paul

adamgarbo, 2022-10-23:
I suspect this issue can be closed now, especially with the imminent release of the OpenLog ESP32. The useof SDIO will likely make this a non-issue.

closed 2022-10-23 by adamgarbo