sparkfun/OpenLog_Artemis_GNSS_Logger #16: 'I2C_BUFFER_LENGTH' was not declared in this scope

adamgarbo, 2020-12-03:
Howdy,

Version 1.8.7 of the SparkFun u-blox library changed the way the I2C transaction size is handled (moved from platform guards to global variable). This causes compilation of the GNSS Logger firmware to fail, as I2C_BUFFER_LENGTH is no longer defined. 

https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/6979cecfce5c4acd3abad1c889e200d63c203b85/Firmware/OpenLog_Artemis_GNSS_Logging/storeData.ino#L183-L184

Would explicitly setting the size (to 32) be the simplest fix? 

Cheers,
Adam

PaulZC, 2020-12-03:
Thanks for spotting this Adam,

Oh, what a tangled web we weave...

I think the _correct_ way to fix this will be to call ```getI2CTransactionSize``` and use that to set ```bytesToRead```?

I'll get on it...!

All the best,

Paul

nseidle, 2020-12-03:
I broke it. Sorry Paul! Thanks Adam for reporting.

PaulZC, 2020-12-07:
Fixed in [this commit](https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/commit/a2ea69b90f841d70d0126b9c80f2efe3b87ac959).
I'll leave this open until v1.3 is released.

adamgarbo, 2020-12-08:
Thanks, @PaulZC!

A nice and simple futureproof fix. That is, at least until the next time Nathan breaks something! 😉 

Cheers,
Adam

PaulZC, 2020-12-08:
v1.3 has just gone live. Please kick the tyres when time permits!
All the best,
Paul

closed 2020-12-08 by PaulZC