sparkfun/OpenLog_Artemis_GNSS_Logger #18: ZED-F9P not recognized by OLA as sensor

scott-mikus, 2021-07-16:
### Subject of the issue
[u-Blox ZED-F9P](https://www.sparkfun.com/products/15136) connected using Qwiic to the OpenLog Artemis does not present as a recognized sensor and no logging begins.

### Your workbench
* 64Gb SD formatted FAT32 which works correctly when used with a ZED-F9R.
* Logging at 1Hz.
* Artemis OpenLog GNSS v1.2
* OpenLog Artemis is wired via a Qwiic cable to both the GNSS and OLA.
* ZED-F9P is not functioning correctly. ZED-F9R works fine.
* OLA is powered via USB-C and Li-ion single cell 3.7v 10,000 mah battery, ZED-F9P and ZED-F9R are powered via their GND and  3v3 and 5v ports along with USB-C and via the OLA module for testing. The ZED-F9R works when being powered from all options. ZED-F9P powers up on all conditions but none offer logging.
* Both GNSS are setup to provide UBX data on their I2C ports and have their other UART2 and SPI ports disabled to save processor and battery power. However they have been tested with all ports providing UBX along with NMEA and RTMC. The ZED-F9R will log with nearly any configuration of the ports. However the ZED-F9P always displays on the OLA as no device connected on Qwiic port. Which you can see below when you enter the configuration menu you and select option 2 and attempt to configure the GNSS sensor.

[image]

### Steps to reproduce
To reproduce I simply plug in the GNSS (ZED-F9P) to the OLA via the Qwiic cable and then plug in the Li-ion battery in to the JST port and click the RST button. At this point I receive this message.

[image]

However if I follow the exact same procedure with the ZED-F9R I begin logging as expected as follows.

[image]

### Expected behavior
Again expected behavior is as follows and is what I get when I connect the ZED-F9R.

[image]

### Actual behavior
But this is what happens when I plug in the ZED-F9P.

[image]

PaulZC, 2021-07-16:
Hi @scott-mikus ,
Thanks for reporting this.
Just to make you aware, the GNSS Logger code was written some time ago using v1.2.1 of the Apollo3 (Artemis) core. I haven't tried it using the most recent core: v2.1 (MBED). If you are uploading the pre-compiled binary then all should be well. If you are compiling the code yourself, please make sure you are using v1.2.1 of the core.
Also, the code was written using the older SparkFun Ublox GPS library. That library is now deprecated and the GNSS Logger code has not yet been updated to use the new SparkFun u-blox GNSS library.
But neither of those things would explain why the ZED-F9R works and the ZED-F9P does not.
Some things to check:
How do you have the I2C pull-up split pads configured on both boards? The u-blox modules work best on I2C when the extra pull-up resistors are disabled (split pads open). The modules have internal resistors which work perfectly.
Have you tried resetting the ZED-F9P via u-center? If you have disabled the UBX protocol-in on the I2C port, that would explain what you are seeing. To check: open messages view and select UBX\CFG\PRT. Check that the Target 0-I2C Protocol In is UBX or UBX+NMEA+RTCM3. It helps if you disable NMEA output by setting Protocol Out to UBX (only). You can reset the module completely using UBX\CFG\CFG: click "Revert to default configuration" then click Send. Then click "Save current configuration" and click Send again.
Do you have another way of checking the Qwiic connection on your ZED-F9P? Have you tried the standard OpenLog Artemis firmware? Does that detect the ZED-F9P (and the ZED-F9R)?
Let me know how it goes.
Best wishes,
Paul

scott-mikus, 2021-07-16:
Howdy @PaulZC,

Thank you for getting back to me so quickly.
Let me go into a bit more detail to help better explain.
I am currently running the [ZED-F9P](https://www.sparkfun.com/products/15136), [ZED-F9R](https://www.sparkfun.com/products/16344) and [OpenLog Artemis](https://www.sparkfun.com/products/16832) all from Sparkfun. Interlinked using standard Qwiic cables, which are tested and in working order. I am very familiar with the u-Blox ecosystem and it's GNSS receivers along with their GSM cellular products. However, I have just recently entered the Sparkfun domain because of the Qwiic system and its interoperability.

 As far as your points, I am currently running only pre-complied binaries as this is my first steps into Artemis and I want to get a feel before I start getting dangerous with it. But I have successfully used the default firmware to log both the ZED-F9R and ZED-F9P using the OLA with zero issues and I mean zero. As the product description stated, nearly plug and play, which worked perfectly. I also purchased a [Redboard Qwiic](https://www.sparkfun.com/products/15123) along with a basic [Qwiic OpenLog](https://www.sparkfun.com/products/15164) and used that to verify Qwiic connectivity with both the ZED-F9P and ZED-F9R and both functioned flawlessly. With the ability to log all the parameters I could get my hands on at what ever baud rate I wanted while simultaneously broadcasting RTCM RTK data over the UART2 port via a 915MHz telemetry radio with both units and ran into no issues. I also factory reset both boards as you pointed out and tested with all different protocol settings (UBX only, UBX NEMA Etc.). Both ZED-F9P and ZED-F9R i2C "slave address" are set to "0x42". As far as the I2C split pads are concerned, I have not touched them since receiving the unit as they worked flawlessly with the OLA in standard firmware config and with all other devices. So, I had no reason to split the pads. Finally when uploading the firmware to the OLA I used the "OpenLog_Artemis_GNSS_Logging-V10-v13.bin".

So, as you can see I am at a loss as to why the ZED-F9P is not showing up as it works flawlessly in every other instance including this very OLA just using the default firmware. Which leads me to believe that the GNSS_Logger firmware has an issue with the ZED-F9P. But what that issue is... I am still trying to find out!

Please let me know what you think and if you have any suggestions or questions for me.

Thank you @PaulZC!

adamgarbo, 2021-07-16:
Hi @scott-mikus,

It could be worthwhile to try a more recent example that makes use of the new u-blox library. As @PaulZC mentioned, the GNSS Logger hasn't been updated in some time and I suspect that it will eventually be deprecated.

https://github.com/sparkfun/OpenLog_Artemis/blob/main/Firmware/Test%20Sketches/GNSS_Data_Logging/GNSS_Data_Logging.ino

I'd be curious to know if this example works for you. Based on my experiences, I don't believe the pull-up resistor pads would cause the issues you're describing. That being said, it is still best practice to rely only on the internal u-blox pull-up resistors. If you feel like digging into old GitHub issues, there's a long history of why this is.

Cheers,
Adam

PaulZC, 2021-07-16:
Hi @scott-mikus ,
Very puzzling...
Can you please try hitting a key in the terminal emulator. That _should_ open up the menu. Hit d to open the debug menu and then please enable both major and minor debug messages. Exit the menus completely. Reset the OLA and you _should_ see some helpful extra debug messages. Please post those here - they might help is debug what is going on. (As Adam says, this code is quite old now, so I hope my memory isn't failing me here!) If that doesn't work, I'll see if I can duplicate your issue here.
Cheers for now,
Paul

scott-mikus, 2021-07-16:
Howdy @adamgarbo,

I know this will sound dumb but that link you provided is for a .ino file not a .bin file. Will i need to download that .ino and upload it using Arduino IDE? If so, when compiling I am getting an error that function Twowire is not present. but there is no supporting files or test telling me any details about setup or compiling the test sketch. 

I am familiar with Arduino IDE but not when it comes to Artemis.

adamgarbo, 2021-07-16:
Hi @scott-mikus,

While the OpenLog Artemis doesn't have its own board definition, you can still use it as a regular Arduino (with some caveats) and upload normal .ino programs. In the comments of the example, there's a section about how to upload the example to the OLA. Here's a snippet:

```
  Hardware Connections:
  Connect an antenna to your GNSS board if required.
  Connect your GNSS board to OpenLog Artemis using a Qwiic cable.
  Insert a formatted micro-SD card into the socket on OpenLog Artemis.
  Connect OpenLog Artemis to your computer using a USB-C cable.
  Ensure you have the SparkFun Apollo3 boards installed: http://boardsmanager/All#SparkFun_Apollo3
  This code has been tested using version 1.2.1 of the Apollo3 boards on Arduino IDE 1.8.13.
  Select "SparkFun Artemis ATP" as the board type.
  Press upload to upload the code onto the Artemis.
  Open the Serial Monitor at 115200 baud to see the output.
```

Cheers,
Adam

scott-mikus, 2021-07-16:
Well @PaulZC,

I am going to have to do some digging here. Writing my reply's as of 30 minutes ago, the ZED-F9P and OLA setup did not function.

However, as of now it is running like a charm. I am still looking into what I changed but from what I have looked at so far, nothing changed. Same Qwiic port, same power supplies, I2c pads still intact, port protocol still set to UBX only on I2c. But I still wanted to provide the terminal output and a video capture of u-Center in the mean time which you can find below.

[Video Capture of u-Center](https://youtu.be/5vfEIg2G2I4)

Terminal Output - [ZED-F9P_07-16-2021-085647_OLA_Debug.txt](https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/files/6831725/ZED-F9P_07-16-2021-085647_OLA_Debug.txt)

Once I figure out what changed I will post it here and let everyone know. The screen capture video is uploading to YouTube. So, please be patient if it isn't available right away. But hopefully I can figure out what changed and get right back to everyone!

scott-mikus, 2021-07-16:
I am curious however @adamgarbo since this code is "quite old". Are their plans to update the code? if not, are there other packages available for GNSS logging that are compatible with the Sparkfun and Qwiic ecosystems? Also, does this code support the [MicroMod Data Logger](https://www.sparkfun.com/products/16829) with a [Artemis MCU](https://www.sparkfun.com/products/16401) installed? Again, if not are there plans to include this hardware or is there another package for the MicroMod board that will provide the same functionality?

The reason I ask is because as you are aware I have both the ZED-F9P and ZED-F9R and I would like to log them simultaneously which I am sure the OLA would not be able to handle alone. And because I already have the MicroMod data logger with Artemis MCU along with a couple other MCU's ([Teensy](https://www.sparkfun.com/products/16402), [nRF52840](https://www.sparkfun.com/products/16984)) I wanted to see if these would work or if I needed to purchase a second OLA and be done with it.

I am still reverse troubleshooting the ZED-F9P and OLA issue and will have results shortly. I wanted to ask about this while I dug deeper!

adamgarbo, 2021-07-16:
Hi @scott-mikus,

Good questions! 

At some point, I actually re-wrote the GNSS Logger code for use with the MicroMod Data Logging Carrier Board (MMDLCB) and Artemis Processor. However, I ended up abandoning it in favour of the new u-blox library, which includes some very useful logging functionality. 

I'm also currently working on a project that makes use of the MMDLCB and Artemis Processor along with the u-blox ZED-F9P to measure the daily velocity of glaciers in the Canadian Arctic by logging RAWX/SFRBX messages. The code makes use of some of the code from the data logging examples @PaulZC wrote and works quite well: 
https://github.com/adamgarbo/Cryologger_Glacier_Velocity_Measurement_System

I think that in many ways, the new u-blox library has made the GNSS Logger somewhat obsolete. I find it's much easier now to write code to interface with the u-blox receiver directly, rather than working within the constraints of precompiled firmware. For logging from the F9P and F9R simultaneously, I suspect writing your own code will be required.

Happy to help if I can!

Cheers,
Adam

scott-mikus, 2021-07-17:
Howdy @adamgarbo,

Yes that helps a lot! But just so I make sure I fully understand. You are suggesting that in the future aka nowish. Due to the u-blox gnss Arduino library, it is more realistic or practice to write my code in Arduino IDE in INO format and upload that to the OLA or MMDLCB rather than creating my own firmware or waiting for someone to write it for me. I am familiar with the u-blox gnss library as I have been using it quite extensively to write code for use with other u-blox gnss receivers (MAX-MQ8, M9 series, NEO-6M along with the F9R/F9P) in conjunction with the Mega/Due/MKR GSM Arduino boards and Teensy boards. But I was immediately drawn to this repo because it could capture UBX streams for the F9R/F9P nearly turn key or so I thought.🙃 I hadn't yet put together that you could upload Arduino sketches to the OLA. But now that I am aware of that I will have to figure out how to capture the UBX streams using the u-blox library via the IDE.

Any guidance would be of great help!!￼

PaulZC, 2021-07-17:
Hi @scott-mikus ,

Here's a quick summary of the current options:

If you just want to log position, velocity and time in text format, the standard [OpenLog Artemis firmware](https://github.com/sparkfun/OpenLog_Artemis) will let you do that. The OLA firmware has just had a major upgrade and now runs on version v2.1.0 of the Apollo3 (Artemis) core - which uses the Mbed operating system. This opens up possibilities for accessing the sensor readings via BLE and I'm making a start on that code now.

As we've discussed, the dedicated [GNSS Logger code for the OLA](https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger) is outdated - but still works fine. It uses v1.2 of the Apollo3 core and the deprecated SparkFun Ublox GPS library. I do plan to upgrade that code to v2.1 of the core and the u-blox GNSS library, but I have other things to do first.

The OpenLog Artemis is not a 'standard' Arduino board. When writing code for it, there are extra things you need to do to: enable power for the Qwiic bus; add a pull-up on the SPI CIPO line; enable power for the micro-SD card. I did write a [stand-alone GNSS Data Logging example](https://github.com/sparkfun/OpenLog_Artemis/tree/main/Firmware/Test%20Sketches/GNSS_Data_Logging) for the OLA which does all the the extra configuration for you. That code was written using v1.2 of the Apollo3 core. It won't work with v2.1 of the core as the Qwiic I2C pull-up and power-loss interrupt code needs to be updated as a minimum. I do plan to do that, but again I have other things to do first. 

The SparkFun u-blox library contains its own [data logging examples](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/main/examples/Data_Logging). Those were written with the MicroMod Data Logging Carrier Board and Artemis processor in mind, but they are as generic as possible and are intended to run on any hardware. Those examples use the standard Arduino SD library to access the microSD card. That was intentional to make the examples as generic and portable as possible. If you are logging at high data rates, the Arduino SD library may become a bottleneck. Bill Greiman's SdFat library is a better choice - the OLA-specific examples and firmware all use SdFat. Upgrading the library examples yourself to use SdFat is not difficult.

Then there are the SparkFun RTK products: the [RTK Express](https://www.sparkfun.com/products/18019) and the older [RTK Surveyor](https://www.sparkfun.com/products/17369). Those are based around the ESP32 processor and the ZED-F9P. They are extremely feature-rich, Nate has spent weeks making them cover as many possibilities and permutations as possible. These products will work as stand-alone non-RTK dataloggers but may be overkill for your application.

Lots of options to choose from! Not everything software-wise is as up-to-date as it could be, but we're working on that.

Very best wishes,
Paul

adamgarbo, 2021-07-17:
Great summary, @PaulZC!

scott-mikus, 2021-07-17:
Yes @PaulZC a fantastic overview!

And you hit my project right on the head in fact. I am working with the F9P/F9R for a RTK Base / Rover setup with dead reckoning and a moving baseline project once I acquire a second F9P and second TOP106. The OLA in those instances I want to use for of course data logging for example RTK ECEF position mapping for Fixed-In RAWX data collection in remote locations where having a laptop or desktop or even microcomputer is not practical. But instead have a pop up GNSS, antenna, OLA, power supply (Battery, solar cell depending on duration of capture and if the site will also broadcast 915MHz RTCM corrections once established, solar charger and power mgmt) and data relay potentially (Again the 915MHz RTCM radio or Simply telemetry instead of RTCM, GSM, LoRaWan, BLE, Fiber, etc.) but right now what I am most focused on is capturing the RAWX data in the UBX stream at a fixed antenna location for a Fixed RTK base station but letting the OLA do the capturing rather than u-Center tying up a computer in a semi secluded location. When I can use a low power highly capable device such as the OLA.And use the GNSS, OLA and a battery and be done with it.

I am curious how I could contact you @PaulZC and you @adamgarbo outside of this issue. So we can close the issue once I complete my troubleshooting. 

I am also curious about the glacier project and how that data is collected as I did not see any hardware in the project capable of transmitting data. Only the OLA's MicroSD card for storage. Is that going to be a single site or multiple and what is the thinking behind data collection and aggregation and post processing assuming data collecting is a boots on the ground procedure?👍🏼

adamgarbo, 2021-07-18:
Hi @scott-mikus,

Great to hear of your use cases. I'd be more than happy to chat about my project. Do you have an email I can reach you at?

Cheers,
Adam

scott-mikus, 2021-07-18:
Howdy @adamgarbo,

Yes, you can contact me at scott@northmeadowmedical.com. I have a couple other projects I am currently working on that I would love to incorporate GNSS into as. One being an [ADS-B/UAT](https://en.m.wikipedia.org/wiki/Automatic_Dependent_Surveillance–Broadcast) ground station tracking platform. Which gathers both commercial and private aviation navigation tracking to provide a radar like coverage where radar itself may or may not be present and is fed to both the FAA and private third-party providers after processing to create coverage maps and tracking metrics plus many more features. However providing accurate tracking data and extensive coverage requires precise location data and the current GNSS implementation is lacking to say the least. So I would love to chat about not only your project but one or two of mine as well! Below is one instance of a base station I designed and constructed. Which lives on my roof. Two other projects are a stratum 1 GNSS time server like the [LeoNTP](https://store.uputronics.com/index.php?route=product/product&path=60_70&product_id=92) and a GNSS photography aid for alignment and tracking during night shots for tracking position and alignment relative to objects in the sky or on the ground during dynamic shoots with a mechanically stabilized platform providing the corrections. But we can talk more in depth later! 😁

[image] 

[image]

scott-mikus, 2021-07-19:
@PaulZC / @adamgarbo,

I am not sure who to ask about this but i have two very what i think are simple questions about the OLA_Settings file. That i am hoping you too can help with.

Firstly, when logging the OLA creates log files based on the type of data being logged (Data / Serial) and then increments it based on the "nextDataLogNumber" variable. My question is can you change the file name to a custom name rather than the default "Data / Serial" naming?

Second, there is a variable in the "OLA_GNSS_settings.cfg" called "usLoggingDuration". Which I assume of course is the duration that the OLA will log data continuously before powering down logging. My question is, can you disable this value so the OLA logs continuously and never stops? Or must the variable have a value and if so what is the upper limit? Also, if the variable cannot be set to continuous logging. Would you mind me adding that feature to the repo?

The reason I ask about the second question is because I recently went on a road trip and wanted to log the entire trip in a single .ubx file. But i found out a couple hours into the drive that the OLA had stopped logging and I hadn't noticed. So, I ended up with gaps in my logs and a multitude of files because I had to restart and didn't think to disable incrementation and wasn't aware of the logging time value. 🤦‍♂️ But we live and learn.

Below is a screenshot of the two variables. Green is the file nextDataLog value and red is the logging duration value.

[image]

PaulZC, 2021-07-20:
Hi @scott-mikus ,

Re: the Data / Serial file naming. No, sorry, it is not possible to change that. It is hard coded in several; places in the firmware. Changing that is possible but would involve quite a bit of additional code.

Re: the logging duration. If the sleep duration (usSleepDuration) is set to zero, the logger should never sleep. Code snippet is:

https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/28ba69e7155af8a32a6269722593e0939f258379/Firmware/OpenLog_Artemis_GNSS_Logging/OpenLog_Artemis_GNSS_Logging.ino#L328

useGPIO32ForStopLogging is set to zero too, so accidentally pulling pin 32 low would have had no effect either. Strange. It should have just kept logging. Unless the power glitched off? How did you have it powered? And which messages did you have enabled?

Cheers,
Paul

PaulZC, 2021-07-20:
Hi @scott-mikus ,

Sorry - the penny has just dropped... You're using v1.2 of the GNSS Logger firmware? v1.3 contained some important improvements and corrections (some which @adamgarbo discovered). Please do switch over to v1.3 as soon as you can. Upgrade instructions are [here](https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/main/UPGRADE.md).

Also, I've just added a BETA version of v2.0 of the GNSS Logger. (I had some spare time over the weekend and so decided to start work on the update.) It includes the upgrade to v2.1.0 of the Apollo3 core. I've migrated all of the u-blox interface over to v2 of the u-blox library to take advantage of the callbacks and in-built data logging functionality. The code will also now work with 8-series modules (e.g. ZOE-M8Q), it is no longer limited to the 9-series. You can use exFAT SD cards as well as FAT32. I've added support for the High Navigation Rate data provided by modules like the NEO-M8U. And you can now log NMEA messages too should you want to. It is not yet fully tested but please give it a try if you have time. Please create new, separate Issues for any issues you do find - please don't add them here.

I'd like to close this issue if possible as we have - I think - addressed your original issue with the ZED-F9P?

All the best,
Paul

scott-mikus, 2021-07-20:
Howdy @PaulZC,

I agree, I will close this issue as it is most certainly resolved and i have begun testing v1.3. I truly appreciate all the effort both you and @adamgarbo put into helping me! However, I need to open a new one in regard to the exFat formatting not being recognized. 😁 But again Thank you!

closed 2021-07-20 by scott-mikus

PaulZC, 2021-07-21:
Sorry - I forgot to push the new v2.0 BETA binary. It is in the [binaries folder](https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/tree/main/Binaries) now.

scott-mikus, 2021-07-23:
> Hi @scott-mikus,
>
> While the OpenLog Artemis doesn't have its own board definition, you can still use it as a regular Arduino (with some caveats) and upload normal .ino programs. In the comments of [...]

@adamgarbo I am having a hard time finding the webpage you provided "http://boardsmanager/All#SparkFun_Apollo3" which i assume you meant "http://boardsmanager.com" not just http://boardsmanager". But when I navigate to that site it displays an error saying that the site is not up and running and to come back soon. Also the sub site is no logger available.

Can you help me find the correct link or a different way to get the Arduino boards loaded please?

Below is a copy of the error.

[image]

Thank you!