sparkfun/OpenLog_Artemis_GNSS_Logger #19: OLA Device Not Recognizing exFat Formatted MicroSD Cards

scott-mikus, 2021-07-20:
### Subject of the issue
OLA does not recognize exFat formatted SD cards of any size for logging.

### Your workbench
- SD card formatted as exFat. 16Gb, 32Gb, 64Gb, 128Gb, 256Gb, 512Gb via Windows 10.
- Logging Frequency 1Hz.
- Artemis OpenLog GNSS V10-V1.3
- OLA connected to U-Blox ZED-F9P/F9R via I2C with Qwiic cable/connectors.
- Using U-Blox ZED-F9P/F9R
- OLA powered via 3.7v Li-Ion battery or USB-C or both. GNSS powered via I2C or USB-C or Header Pins or all three.
- FAT32 formatted SD cards work with no issue.

### Steps to reproduce
Format SD card via either File Explorer, Disk Management or Diskpart to exFat and verify that the SD card is correctly formatted and accessible. Eject card and insert card into OLA. Connect GNSS via Qwiic Cable and connect antenna to GNSS. Connect power to GNSS and OLA and boot OLA. Connect USB-C to OLA and connect USB-C to computer and open terminal to monitor. OLA then boots and displays "SD init failed (first attempt). Trying again... SD init failed. Is card present? Formatted? Config file read failed: SD offline SD card offline Datalogging offline".

### Expected behavior
Expected behavior would be that the OLA boots and read the SD card and config file and then begins logging normally.

### Actual behavior
Actual behavior is that the OLA boots and does not load the SD card or config file and throws multiple errors and halts logging as seen below. Attached is a copy of the terminal output from a failed exFat SD card attempt.

SD init failed (first attempt). Trying again...
SD init failed. Is card present? Formatted?
Config file read failed: SD offline
Artemis OpenLog GNSS v1.3
SD card offline
Datalogging offline
storeData: we are not logging. Discarding data!

[ZED-F9P_07-20-2021-114751_OLA_Debug.txt](https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/files/6851107/ZED-F9P_07-20-2021-114751_OLA_Debug.txt)

PaulZC, 2021-07-20:
Hi @scott-mikus ,
Please try formatting the card with the SD Association formatter:
https://www.sdcard.org/downloads/formatter/
Best wishes,
Paul

scott-mikus, 2021-07-21:
Howdy @PaulZC,

I downloaded the SDcard formatter and refreshed all six cards. But still face the same issue. I attached two of the formatting details below for reference.

[image]

[image]

scott-mikus, 2021-07-21:
Howdy @PaulZC,

Updating to V10-V20_BETA has resolved the exFAT SD card Issue. I am now able to use all of my exFAT cards with no issue.

Thank you for working over the weekend on this. I truly appreciate it! I'd also love to speak with you offline if possible too at some point. My email is scott@northmeadowmedical.com.

Again, Thank you!

closed 2021-07-21 by scott-mikus

scott-mikus, 2021-07-21:
@PaulZC Confirmed tested with 1Tb exFat Sandisk Ultra C10 A1 SDXC U1 with no issues on V10-V2.0_BETA.