sparkfun/OpenLog_Artemis_GNSS_Logger #20: Simple Firmware Compilation Error

scott-mikus, 2021-07-23:
### Subject of the issue
I am sure this is the most simple issue and I apologize for being such a pain about this repo but its a fantastic piece of software and I want to participate as much as I can.

The issue I am facing is when compiling the firmware in AIDE (Arduino IDE) with the Sparkfun Redboard Artemis ATP board selected using your release candidate code unaltered. I receive an error stating "no matching function for call to 'TwoWire::TwoWire(const byte&, const byte&)'" 

### Your workbench
- SD card formatted as exFat. 16Gb, 32Gb, 64Gb, 128Gb, 256Gb, 512Gb, 1Tb
- Logging Frequency 1Hz.
- Artemis OpenLog GNSS V10-V1.3
- OLA connected to U-Blox ZED-F9P/F9R via I2C with Qwiic cable/connectors.
- Using U-Blox ZED-F9P/F9R
- OLA powered via 3.7v Li-Ion battery or USB-C or both. GNSS powered via I2C or USB-C or Header Pins or all three.

### Steps to reproduce
Compile release candidate on my computer.😁 

### Expected behavior
Firmware compiles with no errors.

### Actual behavior
Firmware throws "no matching function for call to 'TwoWire::TwoWire(const byte&, const byte&)'"  error and halts compiling.

Two logs are provided below First is a standard output and second is a verbose output.

[Firmware_Output_Standard.txt](https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/files/6870442/Firmware_Output_Standard.txt)

[Firmware_Output_Verbose.txt](https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/files/6870444/Firmware_Output_Verbose.txt)

Again, I'm sure this is incredibly simple and I am sorry for being such a pain but I have tried for 2+ days now. And searched all over for a solution but can not find one, so i am hoping someone can point me in the right direction.

PaulZC, 2021-07-23:
Hi @scott-mikus ,

It looks like you are trying to compile v2.0 of the GNSS logger code but with v1.2.1 of the Apollo3 core? If you are, that combination won’t work. For v2.0 you need to use v2.1.0 of the Apollo3 core.

Best wishes,
Paul

scott-mikus, 2021-07-23:
Howdy @PaulZC,

It appears I am. I added the primary Sparkfun board manager URL but not the additional apollo URL, I updated to version 2.1.1 and compiled successfully.

I told you it would be a dumb question!🤦‍♂️😁

PaulZC, 2021-07-24:
Hi @scott-mikus ,
Thanks for letting me know. I’d recommend v2.1.0 of the core as v2.1.1 contains a small bug fix which unfortunately makes communication with the ZED less reliable.
Cheers,
Paul

closed 2021-07-24 by PaulZC

PaulZC, 2021-08-10:
Hi @scott-mikus ,

I've just released v2.0 of this code - it contains some improvements over the earlier BETA version.
Please give it a try and let me know if you see any gremlins - please open fresh issues for these.

Very best wishes,
Paul