sparkfun/OpenLog_Artemis_GNSS_Logger #22: Terminal output setting failure

scott-mikus, 2021-08-11:
### Subject of the issue
Terminal output can not be modified from the settings menu within the terminal. Terminal output follows the OLA settings.cfg file regardless of what changes are made in the terminal window during use.

### Your workbench
* Are you using a microSD card? All sizes exFAT
* At what frequency are you logging? 1Hz
* What version of firmware are you using? OpenLog_Artemis_GNSS_Logging-V10-v20.bin
* How is OpenLog Artemis wired to your GNSS? Qwiic
* Which GNSS module are you using? ZED-F9P/F9R
* How is everything being powered? Multiple power options, Battery/USBC/Qwiic/Headers

### Steps to reproduce
Enter terminal window and hit any key to enter OLA menu, then select logging and select option to enable terminal output. Enable terminal output then exit and return to logging. If terminal logging is disabled in the config file terminal logging will not start. If it is enabled but you disable it in the menu it will not disable.

### Expected behavior
Expected behavior is for the setting to be changed within the menu during operation.

### Actual behavior
Actual behavior is that the setting follows the config file no matter what you select during operation.

PaulZC, 2021-08-11:
Hi @scott-mikus ,

Good catch - thank you!

I disable the terminal output while the menus are open (but let the logging continue):

https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/620179d48cbce291525b65fd0eab748dd7342f34/Firmware/OpenLog_Artemis_GNSS_Logging/menuMain.ino#L12-L14

I need to pass ```prevTerminalOutput``` to the logging menu and then restore the modified value before recording the settings:

https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/620179d48cbce291525b65fd0eab748dd7342f34/Firmware/OpenLog_Artemis_GNSS_Logging/menuMain.ino#L132-L134

I'll try and get this done today / tomorrow.
All the best,
Paul

PaulZC, 2021-08-11:
Actually, I do do that for the debug settings, but I forgot about the terminal output:

https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/620179d48cbce291525b65fd0eab748dd7342f34/Firmware/OpenLog_Artemis_GNSS_Logging/menuMain.ino#L81

scott-mikus, 2021-08-12:
@PaulZC It was keeping me honest when I was doing some logging. 😁 The OLA was telling me you don't really need to use my extra resources for this... Right? I'll handle the logging, don't you worry your about that. You told me not to show you the output in the CFG file. Why are you making trouble!😁

Take your time sorting it out and I will test it for you once the fix is published.

Thank you @PaulZC

PaulZC, 2021-08-12:
Hi @scott-mikus ,
Fixed in v2.1. Please give it a try.
Best wishes,
Paul

closed 2021-08-12 by PaulZC

scott-mikus, 2021-08-12:
Thank you @PaulZC, I will give it a shake down and let you know if I run into any issues.

scott-mikus, 2021-08-12:
@PaulZC I have loaded v2.1 on three OLA and tested it with both ZED-F9P/F9R for 5 hours now. And can confirm that both the logging menu and the CFG file logging options function as intended. And transition properly when changed during operation.

Thank you!

PaulZC, 2021-08-12:
Hi @scott-mikus ,

You’re very welcome. Please keep the useful feedback coming!

Best wishes,
Paul