sparkfun/OpenLog_Artemis_GNSS_Logger #24: Logging SD ZED-F9P

guiguidudu, 2021-08-31:
I am trying to log data from a ZED-F9P. The card is connected with a QUICK cable. Messages are recorded on the SD when the logger is connected to the computer but not when it is running on battery.
It doesn't work for v20&21 but works for v13.

PaulZC, 2021-08-31:
Hi @guiguidudu ,

When you disconnect the power from the OpenLog Artemis, it will go into deep sleep. It needs a reset to wake it up again. 

Please try:
- Connect to your computer via USB. Check the LEDs to make sure it is logging
- Then connect the battery
- Then disconnect USB
- Does it keep logging? (Do the LEDs continue to flash?)

Please also try:
- Connect to your computer via USB. Check the LEDs to make sure it is logging
- Disconnect USB. Check the LEDs have gone off
- Then connect the battery
- Press the Reset button
- Does it start logging again? (Do the LEDs start to flash?)

Best wishes,
Paul

guiguidudu, 2021-08-31:
I don't have access to the system for a few days.

I have tried option 2 several times. After a reset the LEDs are blinking but nothing on the SD. If the SD is empty it creates the configuration file and the log file *_0.ubx, empty. After a second reset the file numbers do not increment.

PaulZC, 2021-08-31:
Are you compiling the code yourself? Or are you using the pre-compiled binaries? If you are compiling the code yourself, please use v2.1.0 of the Apollo3 core. v2.1.1 has a bug which makes communication with the u-blox modules problematic.

If the log file is empty, the code will try to re-use it. So do not worry if the file names do not increment. We just need to work out why you are not logging any data.

Have you tried removing the pull-up resistors on the ZED-F9P board by opening the I2C split pad jumpers? That can help the I2C communication.

How are you stopping the logging? Are you using a stop-logging button (connected to pin 32)? If you disconnect the power without stopping the logging, the file will be left open and will appear empty. Maybe this is your problem? Please see the [OpenLog Artemis Hook-Up Guide](https://learn.sparkfun.com/tutorials/openlog-artemis-hookup-guide/low-power-considerations) for more suggestions.

guiguidudu, 2021-09-09:
Sorry for not replying, earlier.

I'm using the pre-compiled binaries

I didn't try to remove the pull-up resistors. I will try so.

I'm not using the pin32. I usually reset the module when I start a new record. Did the behaviour change between v13 and v20?

PaulZC, 2021-09-09:
Hi @guiguidudu ,

Using Pin 32 to stop the logging is the only way to guarantee that the microSD file is closed properly. You can configure this in the Power Options menu.

We had to make _many_ changes for v2.0... Looking at the new code again today, I have noticed that I am not calling ```gnssDataFile.sync();``` inside ```storeData```. That could explain why your files appear empty. I also seem to be ignoring ```settings.frequentFileAccessTimestamps``` which is not good either.

Please leave this with me. I will try and release new code soon.

Thank you,
Paul

PaulZC, 2021-09-09:
Hi @guiguidudu ,

Please try the new [v2.2 BETA firmware](https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/tree/main/Binaries). Please let me know if this makes the code behave the same as v1.3.
Please check that "Frequent log file sync and access timestamps" is enabled in the Configure Logging menu.

Best wishes,
Paul

guiguidudu, 2021-09-09:
Hi @PaulZC , 

it seams to work like a charm with a ZED-F9P. I will try later with a SAM-M8Q.

Thank you very much for your fast reply and your kindness.

Kind regards,
-- 
Guillaume

PaulZC, 2021-09-13:
Hi Guillaume (@guiguidudu ),
Thanks again for reporting this issue. I will include the corrections in release v2.2.
Please open a new issue if you notice any more problems.
Good luck with your project!
Best wishes,
Paul

closed 2021-09-13 by PaulZC