sparkfun/OpenLog_Artemis_GNSS_Logger #26: Feature request: full data of UBX messages sent to terminal

aal7, 2022-01-19:
Using the v2.2 binary as downloaded from the repo, NMEA messages are shown in full on the terminal screen but UBX messages are not. NMEA messages can be viewed live in u-center but UBX messages have to be recorded to a file and then played back later. Is it possible for UBX messages to also be shown in full?

I've added a screenshot which illustrates my point:
[image]

Perhaps I've just missed an option somewhere but I'd like to be able to view the UBX messages in u-center live.

Thank you

PaulZC, 2022-01-19:
Hi @aal7 ,

That's a nice suggestion!

The code prints a summary of the type of UBX message received ("NAV-CLOCK") only because I thought nobody would be interested in seeing all the UBX binary data. But being able to pipe it into u-center and have it 'decode' it in real time is: a) a really nice idea; b) something I've never tried!

It might take me a while - I have a ton of other projects on the go at the moment - but I will add this as a feature and make it switchable (defaulting to off).

u-center will be trying to send occasional UBX poll requests to the OpenLog though. That might have some unexpected results if it happens to be the same as the keystrokes required to open a menu...

Alternatively, I could leave the USB Serial as it is now, and instead pipe both NMEA and UBX out of the Serial1 TX Breakout pin. Would that work for you? You'd need an FTDI Breakout or similar to convert the TX data back to USB COM port. Would you be OK with that?

Thanks again for the suggestion,
Paul

aal7, 2022-01-20:
I guess anything that allows me to viw UBX live in u-center is good. The FTDI breakout is certainly an acceptable solution, although not needing another PCB would be nice!

Out of interest, the NMEA I've shown above seems to play live in u-center just fine. What makes the UBX data different?

Many thanks for engaging.

PaulZC, 2022-01-20:
UBX is a binary (not human-readable) protocol. If I copied it raw to the terminal, you could just see a whole mess of control characters. u-center would understand it, but it would look a mess to a human. I could display it in the terminal as ASCII Hex, but I don't think you'd find that useful, and u-center wouldn't then be able to understand it.

The neatest solution is to output both NMEA and UBX on the TX pin, so you can pass it to u-center via an FTDI Breakout. That would leave the terminal display 'clean' for us humans.

aal7, 2022-01-21:
I don't suppose you could translate the "mess of characters" into hex bytes in ascii?  Just a thought, though. Thanks again.

aal7, 2022-01-24:
Or maybe do both? ;-)
Anyway, thanks for whatever you decide to go with.

PaulZC, 2023-01-08:
Hi @aal7 ,

Apologies for the **_very_** slow response on this. (It's almost a year since you requested this! Sorry about that.)

I've added this feature in v3 of the firmware. Please open the logging menu and use options 11-13 to configure UBX and/or NMEA output on the TX pin. It seems to be working nicely - I've got u-center ticking along with UBX data at the same time as logging UBX and NMEA on the OLA.

Please try this BETA version of the firmware. But please note that this will only work with newer GNSS modules that support the Configuration Interface. If you have a MAX-M10S or any of the F9 family, you should be good to go. If you have an M8, I'm afraid you're stuck with v2.2. Sorry about that.

https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/main/Binaries/OpenLog_Artemis_GNSS_Logging-V10-v30_BETA.bin

The upgrade instructions are here:

https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/main/UPGRADE.md

Please let me know if it works OK for you.

Best wishes,
Paul

PaulZC, 2023-01-11:
Resolved in v3.0: https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/releases/tag/v3.0

Closing...

closed 2023-01-11 by PaulZC