sparkfun/OpenLog_Artemis_GNSS_Logger #28: Possible incompatibility with ZED-F9R HPS 1.30

PaulZC, 2023-01-07:


PaulZC, 2023-01-07:
Hello @baegas20 / Dong-Geulami,

Ah! OK! You are using the OLA GNSS Logger firmware. Sorry! I thought you were using the standard OLA firmware.

OK. I understand completely now. Yes, the GNSS Logger firmware will not work with HPS 1.30 - as it uses some messages which are now deprecated (no longer supported). I will need to change the firmware so it uses v3 of our u-blox GNSS library:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3

v3 of the library is public - but it is still "work in progress" and is not yet released. I am still adding and testing examples. As soon as I have added the missing examples, I will update the OLA GNSS Logger. It will take me a few days. Thank you for your patience and support.

Best wishes,
Paul

baegas20, 2023-01-07:
Hello Paul
Thanks for checking, looking forward to the firmware update and hope it works well
thank you

PaulZC, 2023-01-08:
Hello @baegas20 / Dong-Geulami,

Please try this BETA version of the firmware:

https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/main/Binaries/OpenLog_Artemis_GNSS_Logging-V10-v30_BETA.bin

The upgrade instructions are here:

https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/main/UPGRADE.md

Please let me know if it works OK for you.

Thank you,
Paul

baegas20, 2023-01-09:
Hello @PaulZC
Firmware V30 beta works fine
Raw data recording is also good. There are no data errors or interruptions so far. If there is a problem while using it, I will leave a message here.
Thank you sincerely

closed 2023-01-11 by PaulZC