sparkfun/OpenLog_Artemis_GNSS_Logger #30: add UBX-RXM-MEASX for  u-Blox M10

murmurSK, 2024-02-15:
Hello.
Would it be possible to add an option to log UBX-RXM-MEASX messages for u-Blox M10 receiver (https://www.sparkfun.com/products/18037) as the UBX-RXM-RAWX seem to be deprecated.

PaulZC, 2024-02-15:
Hi @murmurSK ,

The M10S is a Standard Precision GNSS module. It does not support RXM-RAWX. Only High Precision GNSS modules like the ZED-F9P support RAWX...

It would be possible to log RXM-MEASX. But, we would need to upgrade this firmware so it uses v3 of the GNSS library, instead of v2.

Do you have one of our [Thing Plus C - ESP32](https://www.sparkfun.com/products/20168) boards? If you do, you could adapt [this library example](https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/main/examples/Data_Logging/DataLoggingExample9_enableUBXlogging/DataLoggingExample9_enableUBXlogging.ino) to log RXM-MEASX.

Best wishes,
Paul

murmurSK, 2024-02-15:
Hello Paul,
Thank you for the fast response.
Unfortunately, I only have the M10 and Artemis boards.
I have already upgraded the logger to v3 firmware

PaulZC, 2024-02-20:
Hi @murmurSK ,

Please try version 3.1 of the firmware. It includes support for RXM-MEASX. Upgrade instructions are [here](https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/main/UPGRADE.md).

Best wishes,
Paul

closed 2024-02-20 by PaulZC