sparkfun/OpenLog_Artemis_GNSS_Logger #33: 2hours log in one file

bugalaws, 2024-05-26:
There are some way to save a log with one file only and more than 6minutes?

PaulZC, 2024-05-27:
Hi @bugalaws ,

Yes. Please open the serial menu, select option 1 "Configure Logging":

* Set option 8 "sleep duration" to zero for continuous logging
* Or set option 7 "logging duration" to the required duration in seconds, and set option 8 to the required sleep duration.
  * The firmware will log for "logging duration" seconds, then sleep for "sleep duration" seconds, then wake and repeat
  * To re-use the same log file, ensure option 9 is set to "no"

Closing. Please reopen if you need more help with this.

Best wishes,
Paul

closed 2024-05-27 by PaulZC

bugalaws, 2024-05-27:
Hi Paul, I think I don’t express exact I need… in the truth I need the ubx file complete, for example one UBX file to 3 or 4 hours of data..

PaulZC, 2024-05-27:
If you set option 8 "sleep duration" to zero, the firmware will keep logging to one file until you tell it to stop.

You can stop logging using option "q" in the main menu.

Or select option 4 "Configure Power Options", then select option 2 "Use pin 32 to Stop Logging". The logging will stop when you pull pin 32 low.

PaulZC, 2024-05-27:
Yes, it is saved both in flash ("EEPROM") and on SD card.