sparkfun/OpenLog_Artemis_GNSS_Logger #34: Possible bus data corruption - NEO-M8P

PaulZC, 2024-10-10:
As reported on the u-blox forum:

https://portal.u-blox.com/s/question/0D5Oj00000Ux8tRKAR/conversion-of-ubx-to-rinex-in-rtklib-fails-sometimes-only-converts-a-few-samples-rawx-and-sfrbx-messages-are-recorded

The user's dataLog file contains no UBX checksum errors:

[image]

But @cturvey has identified [bit errors](https://github.com/rtklibexplorer/RTKLIB/issues/488) in some RXM-RAWX / NAV-PVT UBX packets.

I suspect this could be pull-up related. Has the user opened the I2C jumpers?

lksweber, 2024-10-10:
By opening the I2C jumpers on the Neo-M8P module from SparkFun you mean these tiny little 3 connected contacts next to the I2C bus? I didn't cut them, everything is as in original configuration. Also, in the configurations sent by the OpenLog to the M8P the setting for I2C is: qwiicBusPullUps=0. The qwiic speed is set as qwiicBusMaxSpeed=400000

PaulZC, 2024-10-10:
Hi @lksweber ,

Yes, please open the two jumpers and re-test:

[image]

https://learn.sparkfun.com/tutorials/how-to-work-with-jumper-pads-and-pcb-traces

Thank you,
Paul

lksweber, 2024-10-10:
I am using this firmware
https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/tree/main
in version V10-v22

lksweber, 2024-10-10:
Hello Paul,

I opened the jumper now and performed the same test as yesterday and still having a similar behaviour with lost packets and irregular observations. What other things can I try with this setup to collect the data on to a sd-card, using the qwiic connection of the SparkFun system?

cturvey, 2024-10-10:
There were two example files, at a content level neither had bad checksums. So the firmware was doing its job filtering there. Suspect it was discarding a lot of packet. A metric or measurement of packet loss there might be helpful.
Also a raw log of all data over the wire, saved as large aligned blocks, say 8KB or 32KB, as space affords, would likely be an order of magnitude faster at the card level, and allow for analysis of data integrity on the PC subsequently. 

Saving a large stream of ASCII NMEA data might be more illuminating at the bit 7 sticking would be more apparent, and one might deduce a pattern or at least frequency.

PaulZC, 2024-10-10:
Hi @lksweber ,

OK - thank you for trying that. I will try and replicate your issue. I have hardware here but it will take me some time to set it up.

Best wishes,
Paul

@cturvey : Thanks for the analysis and suggestions!

lksweber, 2024-10-10:
Thank you Paul! If you need any further information please let me know.

Best wishes!

cturvey, 2024-10-10:
I'd honestly prefer if we try to do this in one place rather than fork into endless threads
https://community.sparkfun.com/t/package-loss-of-logging-gnss-data-on-to-sd-card-with-openlog-artemis/61142

The OpenLog Artemis repo has a shed-load of code, you’ll probably need to take some ownership of the code and the mechanics.

The logging to the card seems to be solid, there’s some efficiency and scaling issues, but everything it is writing is done correctly, although corrupt packets with viable checksums are getting through, end-to-end.

The receiver is very solid in terms of data generation. I2C is a frequent cause of headaches. Might be an issue with over-loading the receiver with persistent hammering, if you’ve exhausted content asking again 1ms later, repeatedly, isn’t going to make data that’s 900 ms away arrive any quicker. There’s also clock-stretching, and how the master deals with the slave dragging.

The data read by the logger is already corrupted off the interface, flagging UBX packets with bad checksums, and providing debugging and telemetry on that, might more quickly illustrate what’s going wrong, and when. With a logic analyzer or scope you could use that as a trigger, and inspect the signalling more directly.

Using the UART output might be simpler, the receiver will just keep pushing available data out, and then it’s the loggers job to recover and buffer that without loss. Improving the efficiency of the card writing code to better match the blocks and erases size will help it dwelling unduly long there, potentially stalling the ingress via the UART on the host side.

For USB, perhaps a RPi based board, recording all output directly to files on a card might be worth exploring.

PaulZC, 2024-10-10:
OK. I'm taking a look at the I2C bus data now, to see if I can capture the errors.

I've updated the [SparkFun_u-blox_UBX_HLA](https://github.com/sparkfun/SparkFun_u-blox_UBX_HLA) to help. There was a bug in the way signed values were being decoded in UBX-NAV-POSECEF and UBX-NAV-POSLLH.

PaulZC, 2024-10-10:
Progress:

My setup:
[OpenLog Artemis](https://www.sparkfun.com/products/19426) (without IMU)
SanDisk Ultra 16GB microSD card
[NEO-M8P breakout](https://www.sparkfun.com/products/15005) (micro USB, u.FL)
Connected via Qwiic - NEO is powered via Qwiic only
I2C pull-up jumper links are **open**
[saleae Pro analyzer](https://www.sparkfun.com/products/13196) is monitoring the I2C bus

Messages:
[image]

Rate:
[image]

Bus:
[image]

With the V10-v22 (v2.2) binary loaded, I see some bus errors (invalid checksums):
[image]

Artemis I2C always was a little delicate...

I re-built v2.2 using Apollo3 v2.1.0 and whatever libraries I had installed on my machine, including SparkFun u-blox GNSS library 2.2.27 and Bill Greiman's SdFat 2.2.3. And with that binary, I no longer see the bus errors...

I'll leave the code running for a while. If I still see no errors, I'll upload the binary and you can give it a try.

What's changed in this new binary? To be honest, I'm not certain. One thing is that SparkFun u-blox GNSS library 2.2.27 doesn't keep [setting the I2C 'register' address to 0xFF](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/bfa5d9d2531688522240d89d44df67d344add1c5/src/SparkFun_u-blox_GNSS_Arduino_Library.cpp#L1193-L1205). That's an old fix so I'm kind of surprised it wasn't already in there. But maybe that's it...

PaulZC, 2024-10-10:
@lksweber : please try:

https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/main/Binaries/OpenLog_Artemis_GNSS_Logging-V10-v22_2024-10-10.bin

Upgrade instructions are [here](https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/main/UPGRADE.md).

Best wishes,
Paul

PaulZC, 2024-10-10:
Just to confirm, the original v2.2 firmware _**was**_ doing the unnecessary set-the-register-to-0xFF transactions. Today's version does not...

[image]

cturvey, 2024-10-10:
Yes, the I2C will auto increment 0xFD,0xFE,0xFF[recurring]

Not sure how that would cause occasional bytes to read with bit 7 set.

The analysis/decode of the bus traffic does look interesting/useful.

lksweber, 2024-10-14:
Hello! Thanks a lot for your help. It definetely got better with this fix, but still the messages I wanted to transmit at the rate just don't work over I2C and many got left out. The UBX-MON-TXBUF got instantly full. I will reconfigure my setup to UART communication, fortunately the hardware does also allow it.

PaulZC, 2024-10-14:
Hi @lksweber ,

The issue is probably that UART1 is being overloaded - by the standard NMEA messages at the increased navigation rate.

If you give me some time, I will create another binary where NMEA is disabled on UART1.

Best,
Paul

PaulZC, 2024-10-14:
Hi @lksweber ,

Please try:

https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/blob/main/Binaries/OpenLog_Artemis_GNSS_Logging-V10-v22_2024-10-14.bin

The ```Configure GNSS Device``` menu now has an option to disable NMEA on UART1:

[image]

The default is "No". Change it to "Yes" to disable NMEA on UART1.

If you leave the standard NMEA messages enabled on UART1 and increase the navigation rate - without increasing the baud rate to match - the messages can overload UART1. This in turn causes problems on I2C.

I gave the new binary a quick test and it seems to be keeping up OK at 5Hz. But, the bus is _**busy**_... You may be able to push it a little higher, but not much...

[image]

I hope this helps,
Paul

PaulZC, 2024-10-14:
This is a ~30 minute test at 5Hz. Everything seems OK. No bus errors and no dropped messages - as far as I can tell:

[image]

PaulZC, 2024-10-14:
Thanks Clive,

I'm using Bill Greiman's SdFat to do the writing, and that writes data in 512 byte sectors...

I'm happy with the binaries as they are now. The updated v2.2 binary for M8 platforms seems to have resolved the bus errors. Disabling NMEA on UART1 helps too. I should have included that setting a long time ago...

When I get time, I'll try and distill an example to prove if the unnecessary set-register-address-to-0xFF was causing or contributing to the bus errors.

cturvey, 2024-10-14:
Kind of construct I use for FATFS on embedded
```
// Simple aligned file chunker - sourcer32@gmail.com
//  for more optimal SDIO/SDMMC file writing via FATFS, etc
//
// Pick size layers below support, and you can afford
//  should be a power of 2 multiple of 512-byte sectors.
//
// Saves file system from reads/writes deblocking your content into the
//  medium / file-system expectations.
//
// The cards typically have a floating pool of pre-erased 128KB NAND
//  blocks to accomodate new writes, and writes-in-place. So the size
//  of your write is somewhat disconnected from the time it takes to
//  write them, but you churn retired blocks, that must be erased.
//
// The card is going to want to concatenate linear/consecutive writes
//  within it's own buffers, it understands minimum write units of
//  512-bytes

#define HOLD_SIZE 8192 // or 512 if down-stream supports single sector

// size =  0 ignored
// size = -1 initialize statics
// size = -2 close/flush and rese statics

void buffer_write(int size, uint8_t *buffer)
{
  static uint8_t holding_buffer[HOLD_SIZE];
  static int used = 0;

  if (size > 0) // special?
  {
    if ((used + size) >= HOLD_SIZE) // filling, or exceeding
    {
      while(size) // whilst content
      {
        int take = min((HOLD_SIZE - used), size);

        if (take) // whilst space
        {
          memcpy(&holding_buffer[used], buffer, take);
          buffer += take;
          size -= take;
          used += take;
        }

        if (used == HOLD_SIZE) // whilst full
        {
          gnssDataFile.write(holding_buffer, used);
          used = 0;
        }
      }
    }
    else
    {
      memcpy(&holding_buffer[used], buffer, size); // Take all into buffer
      used += size;
    }
  }
  else // special
  {
    if (size == -1) // open
     used = 0;
    else if (size == -2) // close / flush
    {
      if (used)
      {
        gnssDataFile.write(holding_buffer, used);
        used = 0;
      }
    }
  }
}
```

PaulZC, 2024-10-15:
This is resolved. Please see the notes here:

https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/releases/tag/v3.2

closed 2024-10-15 by PaulZC